jclark/satpulse #72: Provide graphical view of satellites in web app

jclark, 2025-04-08:
The web app (enabled by the http config section) would be a lot more appealing with a graphical view of satellites.  Two main views are common:

- sky view, showing position of visible satellites in the sky
- satellite levels, showing as a bar chart

Most satellite monitoring apps seems to have these.

There are obviously two main parts:

- front end, in the TypeScript code, which receives the SSE event and renders the view
- back end, in the Go code, which handles the message from the receiver, and sends the SSE event

There is lots to think about in each part.

jclark, 2025-04-08:
For the front end, I think we first need to switch away from using my lame handwritten CSS into something that allows us to leverage more expertly written CSS. [Tailwind.css[(https://tailwindcss.com/) looks suitable for this, and I think can be combined with our current Preact approach. I think Prereact can continue to work for us.

jclark, 2025-04-08:
In terms of rendering the sky view, we have two options: a canvas and SVG.  I think SVG will work better. Performance-wise we only need at most one update per second, and this will give us smooth vector graphics.

In terms of rendering the bar chart, I think we can probably just use CSS boxes, without anything more graphical than that.

jclark, 2025-04-08:
For the backend, the starting point is the messages we are going to use. With UBX, there is

- UBX-NAV-SAT, which gives us information for each SV (space vehicle):
   - SV identity, as gnssId and svId (presumably within the GNSS), one byte each
   - position as elevation and azimuth
   - signal strength (CNO = carrier to noise ratio in dBHz)
   - svUsed flag (saying whether it's being used)
 - UBX-NAV-SVINFO, which is similar to UBX-NAV_SAT but only one byte for SV identity (independent of gnssId)
- UBX-NAV-SIG, which gives detailed info on the signals being acquired/tracked, including
   - satellite identity
   - signal identity
   - signal strength (CNO again)

UBX-NAV-SIG is going to be interesting in a multi-band world, but otherwise not so much.

We need to be careful about bandwith. We don't want to automatically enable messages that would exceed the capacity of a 9600bps channel. UBX-NAV-SAT uses roughly 8 + 12*numSVs bytes. Suppose we have 30 SVs. That's already about 3,300 bits (at 8 data bits and 1 stop bit per byte), which is more than a third of our available bandwidth!  UBX-NAV-SIG is going to be even more: it is 8 + 16*numSigs bytes, but there can be several signals per satellite. So I think it would be too risky too enable UBX-NAV-SIG at 9600bps. We also don't know what order these messages will arrive in. Will the UBX-NAV-TIMEGPS arrive before or after the UBX-NAV-SIG? If there's too much of a delay, it could jeopardize reliable association of UBX-NAV-TIMEGPS with the second.

We also need to think about protocol versions. M8T has UBX-NAV-SAT but not UBX-NAV-SIG. So looks like UBX-NAV-SIG came in with 9th gen. Need to figure out exact protocol version. It's in 29.20 but not in M8T. So enable for 9th gen and only with baud rate of 38400 or more (which is default for later versions). So there is also a subtle dependency on satpulsetool since that it what will make it easy to change baud rate.  Looks like LEA-6T uses UBX-NAV-SVINFO.  But this isn't in M8T. So somewhere between the two, there was a switch. 7th gen at protocol version 14 is still using SVINFO. So we should probably switch to UBX-NAV-SAT with first 8th gen.

For NMEA, we have GSV, which has for each satellite:
- SV identity as a single number
- position as elevation and azimuth
- signal strength

There are a couple of complexities with the GSV message
- it's a multiple message, each NMEA message says it is part M or N
- when there are multiple constellations, there will be multiple sets of messages one per constellation

We don't currently have anything for combining NMEA messages into our higher level abstract messages. What's tricky here is that you don't know when you're done. I think we would need to use our timer to detect a gap and detect when we are done.

jclark, 2025-04-24:
We should also be able to display which satellites are used. This requires parsing GSA sentences as well.

geerlingguy, 2025-04-24:
If I'm reading this correctly, looks like it would give something similar to the `pygpsclient` GUI, but through the browser?

[image]

jclark, 2025-04-24:
@geerlingguy It won't be as sophisticated as that. The goal of the web interface is just monitoring: give you a good idea of how well your GPS is working.  I want it to be simple enough that it can work on a phone browser.

I haven't tried pygpsclient, but it looks interesting.  I mostly use u-center. satpulse is intended to allow use of tools like pygpsclient (or u-center) for more complex diagnostics and configuration: you can just put

```
[[proxy.tcp]]
listen=":2006"
```

in satpulse.toml and then connect to  that TCP socket in pygpsclient.

jclark, 2025-04-24:
For example, here is what the signal levels display looks like on my phone:

[image]

geerlingguy, 2025-04-24:
Ah, perfect then!

Thinking on this a little further, can there be some simple API or MQTT integration so you could, say, build a little Home Assistant card with GPS status, or feed some data into Grafana? That may be a longer term goal, but you have me thinking about more things I could do once I have Satpulse going!

jclark, 2025-04-25:
@geerlingguy Grafana support could be interesting. A little research suggests that what is needed for this is a Prometheus /metrics HTTP endpoint. I added #78 for this.

jclark, 2025-04-28:
I tried this with an Allystar TAU1201 using NMEA, and the results were not good.  It was configured (out of the box) to use NMEA 4.00. NMEA 4.00 doesn't know about multiple signals per satellite, so Allystar fakes it by treating each satellite+signal as its own satellite, with a non-standard 3-digit numeric identifier.  The Allystar manual specifies these:

```
GN_NMEA_IDBASE_SBAS (87) 127~141
GN_NMEA_IDBASE_GPS (0) 01~32
GN_NMEA_IDBASE_GPSL1C (GN_NMEA_IDBASE_GPS + 400) 401~432
GN_NMEA_IDBASE_GPSL2CM (GN_NMEA_IDBASE_GPS + 500) 501~532
GN_NMEA_IDBASE_GPSL5 (GN_NMEA_IDBASE_GPS + 650) 651~682
GN_NMEA_IDBASE_GLONASS (64) 65~96
GN_NMEA_IDBASE_GLNG2 (GN_NMEA_IDBASE_GLONASS + 500) 565~596
GN_NMEA_IDBASE_BEIDOU (200) 201~ 263
GN_NMEA_IDBASE_BDSB1C (GN_NMEA_IDBASE_BEIDOU + 400) 601~663
GN_NMEA_IDBASE_BDSB2I (GN_NMEA_IDBASE_BEIDOU + 500) 701~763
GN_NMEA_IDBASE_BDSB3I (GN_NMEA_IDBASE_BEIDOU + 600) 801~863
GN_NMEA_IDBASE_BDSB2A (GN_NMEA_IDBASE_BEIDOU + 650) 851~913
GN_NMEA_IDBASE_GALILEO (300) 301-336
GN_NMEA_IDBASE_GALE5A (GN_NMEA_IDBASE_GALILEO + 650) 951~986
GN_NMEA_IDBASE_QZSS (192) 193~199
GN_NMEA_IDBASE_QZSSL5 (GN_NMEA_IDBASE_QZSS+ 650) 843~849
GN_NMEA_IDBASE_IRNSSL5 (GN_NMEA_IDBASE_GPS + 900) 901~917
```

So as it is now, we will get multiple overwritten label in the skyview, one for each signal emitted by the satellite, plus the numbers make no sense.

jclark, 2025-04-28:
Currently we model things as a set of satellites each with id/azimuth/elevation/signal strength. But this is a simplification: in reality each satellite transmits multiple signals, each with an ID and its own signal strength.  We would have to incorporate this in order to model the information in Allystar NMEA 4.00 messages.

NMEA 4.10 has a signal ID message that we don't do anything with currently. But I think it is probably possible to have the same satellite ID occur in GSV messages multiple times with separate signal IDs. We don't handle this currently.

UBX-NAV-SAT doesn't model separate signal IDs. But there is also UBX-NAV-SIG that provides separate signal strength information for each signal transmitted by a satellite.

Health/quality are attached to the signal rather than the satellite. So really the right model is a satellite has a ID, elevation, azimuth, plus a list of signals, where each signal has an ID, a strength, and some other info.

This would require the UBX code to combine UBX-NAV-SAT and UBX-NAV-SIG into a single message.

But I think we should separate out proper handling a multiple signals and tackle it together with #73.

jclark, 2025-04-28:
Unicore manual has this example:

```
$GPGSV,2,1,08,03,29,287,33,04,14,313,23,32,37,141,44,31,65,039,46,1*63
$GPGSV,2,2,08,25,14,047,30,29,28,068,39,26,72,222,48,16,31,218,36,1*6F
$GPGSV,2,1,08,03,29,287,34,04,14,313,29,32,37,141,40,31,65,039,43,4*6A
$GPGSV,2,2,08,25,14,047,30,29,28,068,33,26,72,222,44,16,31,218,29,4*62
$GPGSV,2,1,05,03,29,287,41,04,14,313,31,32,37,141,46,25,14,047,37,8*6F
$GPGSV,2,2,05,26,72,222,51,8*5F
```

which suggests we can have multiple series of GPGSV each with a separate signal ID. Current code assumes one series, with mixed signal IDs.

jclark, 2025-05-01:
We should also tackle #84 to ensure that the delays introduced from additional messages do not negatively affect correlation of messages with time pulses.

jclark, 2025-05-02:
This is now on the master branch.

What I did was:

- support multiple signals per satellite in the SatellitesMsg data model
- do not yet display this information in the web app
- do not yet generate this for UBX

closed 2025-05-02 by jclark