jclark/satpulse #78: Add  Prometheus-compatible HTTP `/metrics` endpoint

jclark, 2025-04-25:
By adding support for a Prometheus HTTP `/metrics` endpoint, we could enable monitoring of SatPulse to be integrated into Grafana dashboards.

Possible metrics:

- a gauge with two values saying in sync or out of sync
- a gauge with current signed PHC/GPS offset in nanoseconds
- a histogram with the absolute value of nanoseconds
- current value of stats that we currently calculate over user-specified interval (default 30s) and log
- satellite-related statistics
  - number of satellites in view (maybe per constellation)
  - number of satellites in use
  - signal strength

There's a Go library providing a Prometheus client, although it might be simple enough to roll our own.

This would be useful for me personally because I am running a significant number of satpulse instances and this would enable me to monitor them all conveniently.

geerlingguy, 2025-04-25:
Any (or all) of those stats would be amazing! Grafana is immensely helpful in organizing multiple endpoints, I use it at home for Internet/node/power monitoring already. Would be cool to feed timing data in as well.

scottlaird, 2025-05-09:
There's a chrony exporter that exports ~all of Chrony's metrics, but this won't help with PTP/PHC/GPS specific metrics.  It's still enough to do a useful job in tracking performance, though.

closed 2025-07-19 by jclark